marq24/ha-goecharger-api2 #2: "Age of inverter data" linearly increasing until [go-e] Restart Charger device

karliwalti, 2024-05-24:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Three fields of the Diagnostic fields seem to be non-functional. They increase linearly or stepwise over time, despite the system running. These are
sensor.goe_259493_inva
sensor.goe_259493_lmsc
sensor.goe_259493_lpsc

button.goe_259493_rst will be the only way to reset the "counters"

[image]
[image]
[image]

I would expect that sensor.goe_259493_inva and sensor.goe_259493_lpsc oscillate around 5s when PV is running and linearly increase when not. sensor.goe_259493_lmsc should reset at a status change.  

Log below likely also contains other issues, but will report in separate  issue. 

### Add your DEBUG log output

```shell
2024-05-24 13:16:00.269 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -3477.0, 'pPv': 2739.0, 'pAkku': -2396.0}
2024-05-24 13:16:00.269 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -3477.0, "pPv": 2739.0, "pAkku": -2396.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:00.321 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -3477.0, 'pPv': 2739.0, 'pAkku': -2396.0}}
2024-05-24 13:16:00.713 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:00.792 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:00.792 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.079 seconds (success: True)
2024-05-24 13:16:09.021 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -3477.0, 'pPv': 2739.0, 'pAkku': -2396.0}
2024-05-24 13:16:09.022 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -3477.0, "pPv": 2739.0, "pAkku": -2396.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:09.026 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:09.094 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -3477.0, 'pPv': 2739.0, 'pAkku': -2396.0}}
2024-05-24 13:16:09.148 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:09.148 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.121 seconds (success: True)
2024-05-24 13:16:10.267 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -1999.0, 'pPv': 3611.0, 'pAkku': 0.0}
2024-05-24 13:16:10.268 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -1999.0, "pPv": 3611.0, "pAkku": 0.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:10.314 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -1999.0, 'pPv': 3611.0, 'pAkku': 0.0}}
2024-05-24 13:16:14.610 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:14.666 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:14.666 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.056 seconds (success: True)
2024-05-24 13:16:15.268 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -1999.0, 'pPv': 3611.0, 'pAkku': 0.0}
2024-05-24 13:16:15.268 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -1999.0, "pPv": 3611.0, "pAkku": 0.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:15.311 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -1999.0, 'pPv': 3611.0, 'pAkku': 0.0}}
2024-05-24 13:16:19.611 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:19.664 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:19.664 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.054 seconds (success: True)
2024-05-24 13:16:20.268 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}
2024-05-24 13:16:20.268 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -34.0, "pPv": 2669.0, "pAkku": 0.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:20.309 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}}
2024-05-24 13:16:24.611 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:24.668 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:24.668 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.058 seconds (success: True)
2024-05-24 13:16:25.269 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}
2024-05-24 13:16:25.269 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -34.0, "pPv": 2669.0, "pAkku": 0.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:25.312 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}}
2024-05-24 13:16:29.538 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 63 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:29.590 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 1 -> ['hsa']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:29.590 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'zfocore': 'TRIGGER'}
2024-05-24 13:16:29.591 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:16:29.655 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
2024-05-24 13:16:29.655 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.065 seconds (success: True)
2024-05-24 13:16:30.267 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}
2024-05-24 13:16:30.268 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -34.0, "pPv": 2669.0, "pAkku": 0.0}'} to go-eCharger@192.168.1.81
2024-05-24 13:16:30.312 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -34.0, 'pPv': 2669.0, 'pAkku': 0.0}}
[... cut ...]
2024-05-24 13:18:00.314 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': 102.0, 'pPv': 7244.0, 'pAkku': 238.0}}
2024-05-24 13:18:04.611 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 56 keys from go-eCharger@192.168.1.81
2024-05-24 13:18:04.667 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@192.168.1.81
```

marq24, 2024-05-24:
Hi,

as I tried to explain in your additional discussion https://github.com/marq24/ha-goecharger-api2/discussions/3 the integration provide the data as it was read from the wallbox.

I fully agree with you that it looks like the data the wallbox providing for the three mentioned sensors are not plausible. I have updated the documentation in order to explain the functionality (theory of operation) of this integration a bit more and added information how you can check, if the source of information already providing none-plausible data:

#### 2. In case of implausible data

It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka 'the data is not plausible'. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]

Each sensor of this integration have an API-Key identifier in its entity ID. You can manually request values from your wallbox by using this __API key__ via a regular web browser.

E.g. assuming the value of the sensor in question is `sensor.goe_123456_tpa` and your wallbox is reachable via the IP `192.168.22.10`, then you can request/read the 'original' value via the following link (where `tpa` is the API key):

`http://192.168.22.10/api/status?filter=tpa`

so the pattern is:

`http://[wallbox-ip]/api/status?filter=[API-KEY]`

If the plain data that will be returned in such a request is matching the data displayed by the integration, then I would kindly ask t get in contact with go-eCharger, since in such a case the integration is just the 'messanger'.

karliwalti, 2024-05-24:
Indeed, its just the values that are reported by the device.

closed 2024-05-24 by karliwalti