marq24/ha-goecharger-api2 #4: Preprocess charger timestamps obtained from goecharger

karliwalti, 2024-05-24:
### Description

The timestamps obtained fro the API are not very useful in HA. Therefore it would make sense to provide directly the absolute values and hide the raw values.

The math to do is simple 
last reset - timestamp
and is based on trustworthy info from https://github.com/goecharger/go-eCharger-API-v2/issues/177#issuecomment-1742868661

so for example
`{states('sensor.goe_xxxx_rbt') - states('sensor.goe_xxxx_inva')} `

This is a follow-up and somewhat solution to #2 

### API Keys

```shell
scaa
lbp
lccfi
lccfc
lcctc
lmsc
lpsc
inva
lfspt
```

marq24, 2024-05-25:
addressed with - https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.5.7

please be so kind and close this issue when you consider this as solved - TIA

karliwalti, 2024-05-26:
Installed and tested. works like a charm! Thank you.

closed 2024-05-26 by karliwalti