marq24/ha-goecharger-api2 #5: Provide Switch to allow charging

JakobTewes, 2024-06-11:
### Description

With my old hw v1 Go-E Charger, there was a switch entity to enable/disable charging.
Would be great, having an entity, that can control charging.

Thanks and regards

Jakob

JakobTewes, 2024-06-11:
[This integration](https://github.com/syssi/homeassistant-goecharger-mqtt) provides a switch based on field "frc". 
Pls [see here](https://github.com/syssi/homeassistant-goecharger-mqtt?tab=readme-ov-file#buttons)

marq24, 2024-06-12:
The 'frc' entitie is part of this integration - it's a selection - if it does not appear in our setup you shoul provide DEBUG log information

JakobTewes, 2024-06-12:
Can confirm. The sensors now shine up...for my Gemini Flex this took overnight ;-(

closed 2024-06-12 by marq24