marq24/ha-goecharger-api2 #8: Is it possible to connect through the Cloud API?

acamas, 2024-06-28:
### Description

First, many thanks for your work - I think going through the API makes a lot of sense.

I‘d like to invest into a Wallbox that I can connect to HA. However, I’m struggling as I rented a parking spot in a garage a few meters from my apartment, and I couldn‘t connect to the Wallbox via WiFi. This adds the restriction that I need a SIM and a HA integration connecting to the cloud. „Wallbox“ doesn’t have the German EnWG14a certification at the moment.

The cloud API seems to be quite similar to the local API, so I was wondering if it was possible to connect with your integration but using the Cloud API?

https://github.com/goecharger/go-eCharger-API-v2/blob/main/cloudapi-de.md

acamas, 2024-06-28:
Ok, thanks for checking and considering anyways!

marq24, 2024-06-30:
@acamas here you go -> https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.6.4

update to the latest release...

acamas, 2024-06-30:
Wow, many thanks! This just triggered the purchase decision for a go-e wallbox. I’ll definitely try your API and give feedback!

closed 2024-06-30 by acamas