marq24/ha-goecharger-api2 #11: FYI: troubles in 56.8 go-echarger firmware

try2codesecure, 2024-07-13:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [ ] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I'm running this integration with the latest home assistant and an go-eCharger V4, firmware 56.2.
But with go-e firmware 56.8 the integration stop working & didn't start loading (in my case).
It seems there are some api change in the go-echarger firmware.

For more information:
https://github.com/goecharger/go-eCharger-API-v2/issues/231
https://github.com/evcc-io/evcc/discussions/14752
https://github.com/evcc-io/evcc/discussions/9279#discussioncomment-9980568

marq24, 2024-07-13:
From what I read in the different links you shared there are no changes in the API - how I read it, it's new Firmware (of the Wallbox) that needs a "fix"... 

So for now don't use __56.8__ since for me looks like that multiple things (like evcc) does not work correctly...

erxbout, 2024-07-14:
The linked issue on our side describes a bug where the homeassistant mqtt autodiscovery topics are always written when having the newest firmware..

This should not affect any other function of the charger, if so I would need more details..

marq24, 2024-07-14:
@erxbout 

> The linked issue on our side describes a bug where the homeassistant mqtt autodiscovery topics are always written when having the newest firmware..

thanks for this clarification - since I am not so deep into the mqtt stuff I did not followed this in detail - now I understand, that the issue of the mqtt based integration is not related to the report of @try2codesecure with this local polling based one - check...

> This should not affect any other function of the charger, if so I would need more details...

I am wondering more about the post from steve0564 @ evcc [https://github.com/evcc-io/evcc/discussions/14752] - if this could be related? In any case as soon as I am able to test this locally I will come back to your go-eCharger github repro - right now the car is fully charged and it will take same days before re-charging will be required. But since 'this' integration here is just accessing the documented API's I am currently a bit clueless.

marq24, 2024-07-14:
@try2codesecure you might have some more details for us what you mean with `the integration stops working & didn't start loading`... is this related to the pv-surplus automation? Or is it, that in the HA GUI you start the charging - then the question via which controls? (sensor.id's)

try2codesecure, 2024-07-14:
Sure.
Im running the integraton with the Huawei sensors (SUN/LUNA2000).

`service: goecharger_api2.set_pv_data
data:
  pgrid: "{{states('sensor.power_meter_wirkleistung')|float*-1}}"
`

Trace:
`Ausgeführt: 14. Juli 2024 um 10:45:20
Ergebnis:

params:
  domain: goecharger_api2
  service: set_pv_data
  service_data:
    pgrid: -3247
  target: {}
running_script: false
`

It works good with the 56.1 & 56.2 go-e firmware. But after the upgrade to 56.8 it doesn't start the loading in the ECO mode anymore. There are no other changes between.
Currently my car is loading, I'll try to recreate the error again until next weekend with FW 56.8.

try2codesecure, 2024-07-14:
To clarify, i also think that the problem is not related to your integration.
I just want to link the problem that other integration users avoid those "pitfalls".

marq24, 2024-07-14:
ok - so we are talking then about the pv-surplus charging feature (without additional go-eController) - just providing the pgrid via HA automation...

Not sure if you have noticed, that in the past week I have updated (in the docs) the info [acp added]: 

### Do not forget this important settings
Once you have enabled the automation, you also need to:

- Select the 'logic mode': 'Awattar [Eco]' [API-Key 'lmo']
[Logik/Modus: ECO-Modus]

- enable the 'Use PV surplus' [API-Key 'fup']
[Mit PV-Überschuss laden]

- enable the 'Allow Charge Pause (car compatibility)' [API-Key 'acp']
[Ladepausen zulassen (Fahrzeug Kompatibilität)]

in the setting of your go-eCharger - this can be done via the integration!

karliwalti, 2024-07-14:
I can confirm to have also issues after upgrading to firmware 56.8 and PV surplus charging.

In my setup, the charger caps the current on each phase to max 5A.  When there was enough power from PV, I did do the phase switch to 3 phases, but still only 5A on each phase. Downgrade to 56.1 did work again, but unfortunately the sun was then soon gone. 

You can nicely see how PV did not go into car:
[image]

How L1 was limited to 5A (I did downgrade FW  around 14:30 )
[image]

Go-e did see the surplus (so API transmits correct values), but charger did nothing:
[image]

Settings were correct:
[image]
[image]
[image]

There have been no errors reported. 

It does not seem to be linked to the APi2 but more likely to some firmware changes that mess with the PV surplus charging algorithm. Wloud be great to have access to the FW changelog. The app does not show one for 56.2 and 56.8. 
The weirdest thing is that the API can only limit current to minimum of 6A number.goe_259493_ama  (and mine was set to 16A), so not sure how a cap of 5A is even possible.

marq24, 2024-07-15:
@karliwalti 

On the one hand it might be nice for other users of this integration to get the confirmation, that something after the firmware update stopped working, on the other hand side we can't expect, that anybody from go-e will follow here this issue. So the information you have provided should (IMHO) provided directly to go-e (creating there an issue) - typically they reply quite fast.

What I am wondering about is, that if there is an issue with PV surplus charging with the new firmware, then I would expect that all users (even go-eController owners) should have the same problems (like you just described).

mr-gosh, 2024-07-17:
Should all of this also work with to go-e chargers connected? Just wondering if I should open a bugreport if the ha-automation works only for one of the two chargers with the same settings.

marq24, 2024-07-17:
@mr-gosh your question is IMHO a bit "off topic" here... When you have two go-eChargers and have configured both in HA, and when you then use this integration to send your GRID-export data to both instances I would assume, that then __both__ chargers will try to consume this available energy... 

IMHO (but I could be totally wrong) you have to distribute the available power "manually" to Charger1 and Charger2 (e.g. by simply divide the sensor data by 2)... Obviously only if both chargers have cars connected. I would kindly ask you to start a new discussion for your "dual" use case. - TIA

karliwalti, 2024-08-28:
@marq24 The issue with firmware 56.8 seems to exist no longer. From my perspective, the issue could be closed

closed 2024-08-28 by marq24