marq24/ha-goecharger-api2 #12: Add cll to

karliwalti, 2024-07-15:
### Description

`cll` seem to provide interesting information about current limits.
This seems to be a verbose field, so not sure how easy it would be to integrate in HA ?

### API Keys

```shell
cll
https://github.com/goecharger/go-eCharger-API-v2/issues/188
```

marq24, 2024-07-15:
Hi Walter,

I have to check, if & how the values of this field are translated (or need translations)... and where I could find them (the translations + keys).

Have that said I have also noticed that beside cll there are 98 other "new" API keys that are some sort of documented... probably I will not add all of them - but some of these might be useful (just like cll)...

```
    # avgfhz | R | TYPE | Status | Stromnetz average frequency (~50Hz)
    # awpl | W | TYPE | Status | awattar price list, timestamps are measured in unix-time, seconds since 1970
    # bar | R/W | TYPE | Config | Button Allow WiFi AP reset (0=AlwaysLock, 1=LockWhenCarIsConnected, 2=LockWhenCarIsCharging, 3=NeverLock)
    # ccd | R | TYPE | Status | Connected controller data cch | R/W | TYPE | Config | color_charging, format: #RRGGBB
    # cle | R | TYPE | Status | Cloud last error
    # clea | R | TYPE | Status | Cloud last error (age)
    # cll | R | TYPE | Status | Current limits list
    # cmmr | R | TYPE | Config | controllerMdnsMaxResults
    # cmp | R | TYPE | Config | controllerMdnsProto
    # cms | R | TYPE | Config | controllerMdnsService
    # cmse | R | TYPE | Config | controllerMdnsScanEnabled, set to false to completely disable any MDNS searches (debugging)
    # csa | R | TYPE | Status | controller scan active
    # ct | R/W | TYPE | Config | car type, free text string (max. 64 characters)
    # ctrls | R | TYPE | Status | Controllers search result
    # data | R | TYPE | Status | grafana token from cloud for app
    # di1 | R/W | TYPE | Config | digital Input 1-phase
    # die | R/W | TYPE | Config | digital Input Enabled
    # dii | R/W | TYPE | Config | digital Input Inverted
    # dll | R | TYPE | Status | download link for app csv export
    # dsrc | R | TYPE | Status | inverter data source
    # gmtr | R/W | TYPE | Config | gridMonitoringTimeReconnection in seconds
    # gsa | R/W | TYPE | Status | gridMonitoring last failure
    # hai | R/W | TYPE | Config | httpApiEnabled (allows /api/status and /api/set requests)
    # hla | R/W | TYPE | Config | httpLegacyApiEnabled (allows /status and /mqtt requests)
    # la1 | R/W | TYPE | Config | limit adapter 1-phase (in A)
    # la3 | R/W | TYPE | Config | limit adapter 3-phase (in A)
    # lbl | R | TYPE | Config | lastButtonHoldLong
    # lcs | R | TYPE | Status | last controller scan timestamp in milliseconds since boot time
    # lopr | R/W | TYPE | Config | load balancing protected
    # lrc | R | TYPE | Status | last rfid card index
    # lri | R | TYPE | Status | last rfid id (only available when sendRfid)
    # lrr | R | TYPE | Status | last rfid read (milliseconds since boot)
    # lto | R | TYPE | Status | local time offset in milliseconds, tab + rbt + lto = local time
    # lwf | R | TYPE | Status | last wifi connect failed (milliseconds since boot)
    # mcc | R | TYPE | Status | MQTT connected
    # mcca | R | TYPE | Status | MQTT connected (age)
    # mce | R/W | TYPE | Config | MQTT enabled
    # mcr | R/W | TYPE | Config | MQTT readonly (don't allow api writes from mqtt broker)
    # mcs | R | TYPE | Status | MQTT started
    # mcu | R/W | TYPE | Config | MQTT broker url
    # men | R/W | TYPE | Config | modbus slave enabled
    # mhe | R/W | TYPE | Config | MQTT enable homeassistant discovery
    # mht | R/W | TYPE | Config | MQTT homeassistant topic prefix (set to null to reset back to the default)
    # mlr | R | TYPE | Status | MQTT last error
    # mlra | R | TYPE | Status | MQTT last error (age)
    # mqcn | R/W | TYPE | Config | MQTT skipCertCommonNameCheck
    # mqg | R/W | TYPE | Config | MQTT useGlobalCaStore
    # mqss | R/W | TYPE | Config | MQTT skipServerVerification
    # msb | R/W | TYPE | Config | modbus slave swap bytes
    # msp | R/W | TYPE | Config | modbus slave port (requires off/on toggle)
    # msr | R/W | TYPE | Config | modbus slave swap registers
    # mtp | R/W | TYPE | Config | MQTT topic prefix (set to null to reset back to the default)
    # ocppao | R/W | TYPE | Status | OCPP AllowOfflineTxForUnknownId
    # ocppcm | R/W | TYPE | Status | OCPP LocalAuthListEnabled
    # ocppcs | R | TYPE | Status | OCPP connector status (0=Available, 1=Preparing, 2=Charging, 3=SuspendedEVSE, 4=SuspendedEV, 5=Finishing, 6=Reserved, 7=Unavailable, 8=Faulted)
    # ocppf | R/W | TYPE | Config | OCPP fallback current
    # ocppla | R/W | TYPE | Status | OCPP LocalAuthListEnabled
    # ocpplo | R/W | TYPE | Status | OCPP LocalAuthorizeOffline
    # ocppti | R/W | TYPE | Status | OCPP transaction id
    # pco | R | TYPE | Config | controllerCloudKey
    # pdi | R/W | TYPE | Config | protect Digital Input
    # pgr | R/W | TYPE | Config | protect Grid Requirements
    # rdbf | R/W | TYPE | Config | randomDelayStartFlexibleTariffCharging in seconds
[... 11 lines cut ...]
    # rdree | R/W | TYPE | Config | randomDelayReconnectionEndsAt (set to null to abort current randomDelayReconnection)
    # rmaf | R/W | TYPE | Config | reconnectionMaximumFrequency in Hz
    # rmav | R/W | TYPE | Config | reconnectionMaximumVoltage in Volt
    # rmif | R/W | TYPE | Config | reconnectionMinimumFrequency in Hz
    # rmiv | R/W | TYPE | Config | reconnectionMinimumVoltage in Volt
    # rsa | R/W | TYPE | Status | rampup started at
    # rsre | R/W | TYPE | Config | rampupAtStartAndReconnectionEnabled
    # rsrr | R/W | TYPE | Config | rampupAtStartAndReconnectionRate in %/s
    # smd | R | TYPE | Status | smart meter data
[... cut ...]
    # ufa | R/W | TYPE | Config | Underfrequency Control activation threshold
    # ufe | R/W | TYPE | Config | Underfrequency Control enabled
    # ufm | R/W | TYPE | Config | Underfrequency Control mode (TypeNominal=0, TypeActual=1)
    # ufs | R/W | TYPE | Config | Underfrequency Control stop frequency
    # wbw | R | TYPE | Config | WiFi Bandwidth (for both AP and STA) WIFI_BW_HT20=1, WIFI_BW_HT40=2
    # wda | R/W | TYPE | Config | disable AccessPoint when cloud is connected
    # wsl | R | TYPE | Status | WiFi STA error messages log
```

marq24, 2024-07-28:
added a first cll "plain list" (with (partial translated keys) version in 2024.7.2

marq24, 2024-08-21:
@karliwalti can we close this?

karliwalti, 2024-08-22:
Yes, this field is informative and useful, it works. thank you!

closed 2024-08-22 by karliwalti