marq24/ha-goecharger-api2 #13: Surplus data with multiple goechargers

mr-gosh, 2024-07-23:
I cerated the automation for both of my goechargers but the data is only recieved by one of them (which one seems to change from reboot to reboot of the goechargers).

So the question is - how to configure that the pv surplus power usage is available to both chargers?
The distribution if both have a car connected could be configured through the "Lastmanegement" probably - but that is another question.

@mr-gosh your question is IMHO a bit "off topic" here... When you have two go-eChargers and have configured both in HA, and when you then use this integration to send your GRID-export data to both instances I would assume, that then __both__ chargers will try to consume this available energy... 

IMHO (but I could be totally wrong) you have to distribute the available power "manually" to Charger1 and Charger2 (e.g. by simply divide the sensor data by 2)... Obviously only if both chargers have cars connected. I would kindly ask you to start a new discussion for your "dual" use case. - TIA

_Originally posted by @marq24 in https://github.com/marq24/ha-goecharger-api2/issues/11#issuecomment-2233684202_

achmet9948, 2024-07-23:
I have the same error. When I add the second wallbox, the data is only sent to it and not to the first wallbox. If I delete the second wallbox, it works again with the first one.

mr-gosh, 2024-07-23:
> I have the same error. When I add the second wallbox, the data is only sent to it and not to the first wallbox. If I delete the second wallbox, it works again with the first one.

How do you check where the data is recieved?

marq24, 2024-07-23:
ok - the issue is, that even with multiple instances of the integration you just have a single service (right now) called

`service: goecharger_api2.set_pv_data`

To be able to support multiple chargers with different services I must adjust the integration - but in a way that the single go-eCharger users will not be affected... This will take me a while. I am planing, that you must use also the serial number of the charger as part of the service name - so something like this `service: goecharger_api2.set_pv_data_XXXXXXX`

Just in case - be aware, that you must take the responsibility of splitting the available PV power to the different instances of your chargers - if you will provide both with the plain "export to grid" value, that both chargers will try to consume  this available power individually (which will result in twice as high power usage).

mr-gosh, 2024-07-24:
> Just in case - be aware, that you must take the responsibility of splitting the available PV power to the different instances of your chargers - if you will provide both with the plain "export to [...]

I think (if I don't get that wrong) the easiest would be to let the load balancing handle that.

So the same values should be available to both chargers and IF both are loading the load balancing should get the available power and split that to both or "n" devices...

marq24, 2024-07-24:
> I think (if I don't get that wrong) the easiest would be to let the load balancing handle that.

would be a surprise for me - but you will see in a short while...

marq24, 2024-07-28:
check release 2024.7.1

mr-gosh, 2024-07-28:
> check release 2024.7.1

Just updated - seems we have to change the automation after the upgrade?

> Die Automatisierung „go-e PV surplus charging brigde“ („automation.go_e_pv_surplus_charging_brigde“) hat eine unbekannte Aktion: „goecharger_api2.set_pv_data“.

> Dieser Fehler verhindert, dass die Automatisierung korrekt ausgeführt wird. Möglicherweise ist diese Aktion nicht mehr verfügbar oder wurde durch einen Tippfehler verursacht.

> Um diesen Fehler zu beheben, bearbeite die Automatisierung und entferne diese Aktion

marq24, 2024-07-28:
yes of course - check also the updated documentation...

mr-gosh, 2024-07-28:
> yes of course - check also the updated documentation...

Perfect - found the entry:

> Having multiple go-eChargers in your HA installation?
....

```
action:
  - service: goecharger_api2.set_pv_data_012345
    ...
```

achmet9948, 2024-07-30:
Hi I tested it: 
One Wallbox is working and the second on is not working: 
Working Box:
[image]

Not Working: 
[image]

marq24, 2024-07-30:
please be so kind and provide additional log output... - I can't guess from the screenshots what could be the possible root cause...

marq24, 2024-07-30:
and for the start - simply try to use/start the both services manually (via the HA-GUI) - in YAML mode...

Service A
```
service: goecharger_api2.set_pv_data_999999
data: {
  pgrid: 0
}
``` 

Service B
```
service: goecharger_api2.set_pv_data_888888
data: {
  pgrid: 0
}
```

marq24, 2024-07-30:
but update before to the 2024.7.2

mr-gosh, 2024-07-30:
On my system it's working exactly like intended - waiting for the 2nd car for testing...

achmet9948, 2024-08-01:
Okay, I configured again. It's working now 💯

closed 2024-08-01 by marq24

mr-gosh, 2024-08-01:
Thanks for completing this feature.
Awesome work!

marq24, 2024-08-05:
@mr-gosh @achmet9948 please note, that with the next release there will be a breaking change concerning the support for multiple chargers - the serial will be removed from the service name (again) and you now must add an additional data attribute `configid` - details will follow in the main readme.

mr-gosh, 2024-08-09:
> the serial will be removed from the service name (again) and you now must add an additional data attribute `configid` - details will follow in the main readme.

I really don't get it how to find the service ID 

> The simples way to find the corresponding ConfigEntryId's of your multiple configured go-eCharger integrations is by using the GUI of the Service, activate the optional selection field, select the charger and then switch (from GUI) to YAML-Mode mode - this will show you the configid you must use.

The only optional selection field i found is in the "devices overview" but when i select a divice with the checkbox there is no possibility to switch to yaml view.... - i am lost - sorry! :(

marq24, 2024-08-10:
[image]

mr-gosh, 2024-08-20:
Thanks for clarifying - probably this should be integrated into the documentation.

The surplus information is now getting into both chargers.

I wanted to drop a line that it now works and the surplus charging worked very well.
It started to charge and stopped in the evening and started again the next morning when the sun was strong enough.
Forgot to take screenshots.

Thanks for your work!

Will see how this works if really 2 cars are charging at the same time.