marq24/ha-goecharger-api2 #14: Duplicate devices

IngmarStein, 2024-08-20:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I only have one go-e charger, but I think one of the latest updates of this integration caused duplicate devices (with no entities) to appear:

[image]

I assume that happened when adding support for multiple chargers. Is the recommended approach to manually clean up the device storage or will there be an update to do this automatically?

marq24, 2024-08-20:
ok... I found the root cause in my code - or better in the git history - I have recently changed the 'identifier' object for the integration - with each of the changes a new DeviceEntry was created by HA (without deleting the old one)...

With the next release (will be available shortly) I will add code, that will delete all orphan DeviceRegistry entries of this integration.

IngmarStein, 2024-08-20:
Oh nice! Then I‘ll hold off on removing the orphan devices manually and report back once the next release is out.

IngmarStein, 2024-08-20:
That worked perfectly, thanks!

closed 2024-08-20 by IngmarStein