marq24/ha-goecharger-api2 #15: {"message":"Invalid handler specified"}

fredericpapaux, 2024-09-16:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hi! Many thx for your job. I'm discovering HA and when I try to integrate I receive the error message {"message":"Invalid handler specified"}. I will try to make a debug log if you needed but I think this error is due to my lake of knowledge in HA.
Can you help me?

HA Core 2023.7.3-22 on Synology

### Add your DEBUG log output

```shell
{"message":"Invalid handler specified"}
```

marq24, 2024-09-16:
I would kindly ask for a complete log output - in order to get some details about this exception - might be that just the HA version is too old - but could be also the case that just the IP/URL is not correct.

fredericpapaux, 2024-09-16:
[home-assistant_2024-09-16T19-11-53.788Z.log](https://github.com/user-attachments/files/17018235/home-assistant_2024-09-16T19-11-53.788Z.log)

Hi! Thnx for your fast response. I hope this log can help to resolve. I see now the actual version is 2024.9.2. My syno is maybe too old.

Best regard

marq24, 2024-09-16:
thanks! -> is it possible for you upgrade your HA version to 2023.8.0 at least?!

fredericpapaux, 2024-09-16:
I dont think so. My syno is DS214 (10 years but always well). I think a ask it to much this time. Did you agree?

marq24, 2024-09-17:
if you like, you can provide another log - so I might be able to share some info, if there could be a short cut ;-)

fredericpapaux, 2024-09-17:
[home-assistant_2024-09-17T20-27-58.554Z.log](https://github.com/user-attachments/files/17034650/home-assistant_2024-09-17T20-27-58.554Z.log)

Maybe that's can help somebody with this same old NAS.
Many thx

marq24, 2024-09-18:
if you still like to experiment a bit

open `custom_integration\goecharger_api2\__init__.py` and search for:
`def _friendly_name_internal(self) -> str | None:` 

and replace with: (so just change the method signature by adding a 'x')
`def _xfriendly_name_internal(self) -> str | None:`

fredericpapaux, 2024-09-19:
No. Problem seems to be the seem. 
thank you for the try. 
I will install my new syno tomorrow. I think it will be better with the virtual HA. 

Best regard

closed 2024-09-20 by marq24