marq24/ha-goecharger-api2 #17: The sensor "[go-e] Energy total" [eto] is not updated

apromix, 2024-10-06:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [ ] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hello.

I have the go-e Charger Gemini flex 22 kW (limited to 11kW) and I'm using this custom integration.
I have noticed that the **[go-e] Energy total** sensor is not automatically updated .

The sensor **[go-e] Power total now** (nrg) shows the current power consuming - no problem here.
[image]

But the  **[go-e] Energy total** (eto) sensor does not update automatically:
[image]
The updates (the three steps in the graph) were only triggered manually by reloading the Custom integration in the menu. So the initialization (start or reload) of the  Custom integration updates the eto sensor.

I have tried the latest go-eCharger Firmware Version: 56.8 and the latest successfully testes go-eCharger Firmware Version 56.2. But I think the problem is not on the the go-eCharger site because the API responses are OK (eto is incrementing):

`$ curl http://wallbox.fritz.box/api/status?filter=nrg,eto`

`{"eto":2379748,"nrg":[227.8500061,225.6799927,227.2299957,3.099999905,7.144000053,7.144000053,7.18200016,1503.525024,1488.874023,1503.936035,0,4496.334961,92.36772919,92.34718323,92.15492249,100]}
`

`$ curl http://wallbox.fritz.box/api/status?filter=nrg,eto`

`{"eto":2379750,"nrg":[227.5399933,225.6799927,227.5399933,3.099999905,7.18200016,7.105999947,7.21999979,1509.689941,1487.687012,1510.004028,0,4507.380859,92.38143921,92.76696014,91.91435242,100]}
`

`$ curl http://wallbox.fritz.box/api/status?filter=nrg,eto`

`{"eto":2380223,"nrg":[226.6100006,224.75,226.3000031,3.099999905,7.144000053,7.06799984,7.144000053,1496.774048,1473.55896,1486.603027,0,4456.936035,92.45617676,92.76225281,91.95367432,100]}
`

I haven't found any relevant DEBUG logs for this problem. Only some requested API keys are not returned:
`$ curl http://wallbox.fritz.box/api/status?filter=ccu,rfb,rst
{}
`

This issue affects the Energy Dashboard, because the hourly sums of energies (grid + PV + Individual devices) is incorrect.

Many thanks for you help, maybe advise. 

### Add your DEBUG log output

```shell
2024-10-06 17:13:09.136 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 57 keys from go-eCharger@http://wallbox.fritz.box
2024-10-06 17:13:09.195 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://wallbox.fritz.box
2024-10-06 17:13:09.196 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.060 seconds (success: True)
```

marq24, 2024-10-06:
strange... (sorry for my initial reply)

what is the sensor `sensor.goe_XXXXX_tpa` showing on your system? And just to double check - we are talking about the `sensor.goe_XXXXXX_nrg_11` sensor - right? 

here is mine... 
[image]

marq24, 2024-10-06:
... and will your eto updates, when you press the `button.goe_XXXXXX_zfocore` ?

apromix, 2024-10-06:
Hi Matthias.

Yes, `[go-e] Power total now` == `sensor.goe_XXXXXX_nrg_11`, which corespondents to actually power delivery to wallbox.

Your `[go-e] Energy total` sensor `sensor.goe_XXXXXX_eto)` works fine.

No, after press the `button.goe_XXXXXX_zfocore` is the `eto` sensor not updated.

marq24, 2024-10-06:
ok - can you please edit:

`custom_components/goecharger_api2/const.py`

search for `eto`

and replace:
```
        key=Tag.ETO.key,
        handle_as_float=True,
        factor=1000,
        suggested_display_precision=2,
        entity_category=EntityCategory.DIAGNOSTIC,
        native_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        icon="mdi:lightning-bolt",
        entity_registry_enabled_default=True
```

with
```
        key=Tag.ETO.key,
        entity_category=EntityCategory.DIAGNOSTIC,
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        suggested_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
        suggested_display_precision=2,
        state_class=SensorStateClass.TOTAL_INCREASING,
        device_class=SensorDeviceClass.ENERGY,
        icon="mdi:lightning-bolt",
        entity_registry_enabled_default=True
```

so basically remove:
```
        handle_as_float=True,
        factor=1000,
        native_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
```
and add:
```
        native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
        suggested_unit_of_measurement=UnitOfEnergy.KILO_WATT_HOUR,
```

and restart HA...

apromix, 2024-10-06:
Matthias, many thanks for your support.

I tried the patch, but without success. The sensor is still not updating.

Just in the background: I checked the communication between HA and go-e with `tcpdump` and I see that the `eto` is requested and answered with the correct value.

marq24, 2024-10-07:
mhhh - can you please be so kind and use a differnt browser or incognito mode in order that we do not have here a ‘unexplainable’ caching issue? Just few days ago in another integration a user habe reported also an issue that was not reproducable for me (notbpossible to change units) that turned out to be a local browser issue…

marq24, 2024-10-07:
Und noch ne Frage .. hast Du mal versucht eine zweite HA Instanz (im DockerContainer) aufzusetzen... Und da die Integration ganz frisch installiert... Mir gehen irgendwie die Ideen aus...

marq24, 2024-10-07:
... and if its not a Browser issue - would it be possible for you to setup a second (local) HA instance (e.g. via Docker) - I am wondering if there is something internally messed up with the entity id... I do not understand at all WHY the entity only updates its value on Integration start?

apromix, 2024-10-07:
Hallo Matthias,
ja, ich habe schon versucht, leere HA Instanz zu starten, um das Problem zu reproduzieren.  Da hat alles funktioniert.
Gerade habe ich go-e Device aus meiner HA-Instanz gelöscht, die Custom Integration entfernt und die komplette Installation erneut durchgeführt.

Jetzt funktioniert alles wie erwartet. Also k.A. ..

Vielen Dank für Deine Mühe, und dass du so viel für OpenSource Community machst.
TOP Arbeit!

Gruß ..

apromix, 2024-10-07:
Hallo Matthias,

sieht das so aus, dass ich den BUG reproduzieren kann.

Nach der Installation und als ist neu go-eCharger hinzugefügt habe, sind by default nicht alle Sensoren `enabled`.
Wenn ich den Sensor `goe_XXXXXX_ccw_ssid` (**Connected WiFi**) aktiviere, aktualisieren von ETO ist gestoppt, und ich sehe folgende DEBUG Logs:

`ValueError: Sensor sensor.goe_XXXXXX_ccw_ssid has device class 'None', state class 'measurement' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'MySSID' (<class 'str'>)`

**Full logs:**

`2024-10-07 11:52:50.679 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 57 keys from go-eCharger@http://wallbox.fritz.box
2024-10-07 11:52:50.744 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://wallbox.fritz.box
2024-10-07 11:52:50.745 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.066 seconds (success: True)
2024-10-07 11:52:50.749 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 657, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'MySSID'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 660, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'MySSID'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/config/custom_components/goecharger_api2/sensor.py", line 49, in state
    return SensorEntity.state.fget(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 664, in state
    raise ValueError(
ValueError: Sensor sensor.goe_XXXXXX_ccw_ssid has device class 'None', state class 'measurement' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'MySSID' (<class 'str'>)
2024-10-07 11:52:55.751 ERROR (MainThread) [frontend.js.modern.202410022] Uncaught error from Chrome 129.0.0.0 on Linux x86_64
TypeError: Cannot read properties of undefined (reading 'friendly_name')
friendly_name (src/common/entity/compute_state_name.ts:8:13)
entityId (src/data/history.ts:319:6)
computeGroupKey (src/data/history.ts:510:8)
Date(hass.states[entity].last_updated).getTime (src/data/history.ts:453:33)
this (src/panels/history/ha-panel-history.ts:479:10)
callbackFunction (src/data/history.ts:142:17)
callback (node_modules/home-assistant-js-websocket/dist/connection.js:23:33)
forEach (node_modules/home-assistant-js-websocket/dist/connection.js:15:25)`

marq24, 2024-10-07:
Thanks for all the details... as quick fix/hack:

ok - can you please edit:

`custom_components/goecharger_api2/const.py`

search for `ssid`

and replace:
```
        key=Tag.CCW.key,
        idx="ssid",
        entity_category=EntityCategory.DIAGNOSTIC,
        native_unit_of_measurement=None,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=None,
        icon="mdi:wifi",
        entity_registry_enabled_default=False
```

with
```
        key=Tag.CCW.key,
        idx="ssid",
        entity_category=EntityCategory.DIAGNOSTIC,
        native_unit_of_measurement=None,
        state_class=None,
        device_class=None,
        icon="mdi:wifi",
        entity_registry_enabled_default=False
```

so basically replace:
`SensorStateClass.MEASUREMENT`  with `None` 

and restart HA...

marq24, 2024-10-09:
Please update to https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.10.0 and close this issue when this is finally fixed (keep my fingers crossed)!

apromix, 2024-10-09:
:+1:
I will test it tomorrow.
Thanks ..

closed 2024-10-09 by apromix