marq24/ha-goecharger-api2 #28: CARDS Array

dervitti, 2024-11-24:
### Description

Hi,
I would really like to access the information that the API provides in the CARDS array.
Is there any plan to integrate this?
best regards

### API Keys

```shell
CARDS
```

marq24, 2024-11-24:
Mahlzeit! - if you could explain me a 'reasonable' use case (why the info from the CARD's array can generate any value) - then I would considder adding it - right now the info is used for the values in the comboxbox 'trx'

dervitti, 2024-11-24:
Mahlzeit! Thank you for your quick response! ;) The Use Case is: My Wife and me hat a token to unlock the charger, and my son-in-law too. I charge my son-in-law for the kilowatt hours that he charged an my wallbox.

In my old Iobroker, i had a script for that:

var kWhPreis = 0.3697;

on({id: 'go-e.0.unlocked_by', change: "ne"}, async function (obj) {
  var value = obj.state.val;
  var oldValue = obj.oldState.val;
  
  //console.log(['Unlocked by ',value,' Oldvalue: ',oldValue].join(''));
  

  Jetzt = (new Date().getTime());
  var lastchanged = obj.state.lc;
  var res = Math.abs(Jetzt - lastchanged) / 1000;
  
  // wenn die letzte Änderung nicht mehr als 2 Minuten alt ist
  if (res < 120) {
    // Manuel hat entsperrt //oder hatte entsperrt
    if (value == '4' )//|| oldValue == '4') 
    {
      var ZaehlerStandKona = getState("go-e.0.rfid.badges.4.consumption").val
      var ZaehlerWertKona  = ( (ZaehlerStandKona - 123.5)* kWhPreis + 45.65 ).toFixed(2);
      // Sekunden seit lezter Änderung
      Jetzt = (new Date().getTime());
      var lastchanged = obj.state.lc;
      var res = Math.abs(Jetzt - lastchanged) / 1000;

    
      var messagetext = (['Aktueller Zählerstand:\n',ZaehlerStandKona,' kWh, ',ZaehlerWertKona,' € \n'].join('')); 
      if (obj.oldState.val == '4') {messagetext+='Laden beendet'}
      console.log(['Unlocked by ',getState("go-e.0.unlocked_by").val,' Telegram Messagetext: ',messagetext].join('')); 
      sendTo('telegram.1', {text: messagetext}, function (res) {
          console.log('Sent to ' + res + ' users');
      });
        // Wenn Manuel lädt dann auf maximale Leistung stellen
        if (getState("go-e.0.unlocked_by").val == '4') {
            setState("go-e.0.amperePV", 16 , false);
        }

    } 
  };
});

marq24, 2024-11-25:
Thanks for your use-case - I have to check (mainly internal motivation) if I would assist you with the external/additional REST Sensor setup - or if I will integrate energy counter's per CARD... week just started...

dervitti, 2024-11-25:
```
sensor:
  - platform: rest
    name: Go-eCharger Manuel Energy
    resource: http://<IP_DES_CHARGERS>/api/status
    method: GET
    headers:
      Content-Type: application/json
    value_template: >
      {{
        value_json.cards | selectattr("name", "equalto", "<NAME>")
        | map(attribute="energy") | first | default(0) | int
      }}
    unit_of_measurement: "Wh"
```
Do it so far, but it would certainly still be great if one could find the "cards" in your HA-integration. Even with the increasing popularity of electric cars, I can imagine more “guest” access to which you can offer PV electricity while having coffee and cake. 

I suspect it's not as easy with this array as it is with singleValues but hey, I can program a little, but I've found that my skills aren't enough to support you here.

marq24, 2024-11-26:
check the https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.11.1

dervitti, 2024-11-26:
Works perfectly! 💯 Many thanks for your very quick response and support!

closed 2024-11-26 by dervitti

marq24, 2024-11-26:
JFYI: You 'son-in-law' already contacted me in order to tweak the numbers a bit... and of course I get a little share of his savings :-D -> Looks like a cool business model of the future for me...

dervitti, 2024-11-26:
I was suspicious of him right from the start when my daughter brought him along!