marq24/ha-goecharger-api2 #29: Error while setting up goecharger_api2 platform for number

bac83, 2024-11-29:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

After upgrading the intergration to release 2024.11.1 the number entities (like number.goe_267854_awp) won't load.

### Add your DEBUG log output

```shell
2024-11-29 11:59:47.308 ERROR (MainThread) [homeassistant.components.number] Error while setting up goecharger_api2 platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 365, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/goecharger_api2/number.py", line 20, in async_setup_entry
    entity = GoeChargerNumber(coordinator, description)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/goecharger_api2/number.py", line 33, in __init__
    description.native_max_value = 16
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 4, in __setattr__
dataclasses.FrozenInstanceError: cannot assign to field 'native_max_value'
```

marq24, 2024-11-29:
thanks for your report - will fix that ASAP!

marq24, 2024-11-29:
please update to https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.11.2 and close if fixed

bac83, 2024-12-02:
The updated fixed the issue. Thanks for the quick help

closed 2024-12-02 by bac83