marq24/ha-goecharger-api2 #30: API Wrong Unit

max-at, 2024-11-29:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

API Key "cdi" laut Doku von go-E:
`charging duration info (null=no charging in progress, type=0 counter going up, type=1 duration in ms)`
die Dauer wird in Milisekungen ausgegeben, in HA wird bei mir aber als einheit Minuten angezeigt, das gehört korrigiert :-)

[https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md](url)

[image]

lg max

marq24, 2024-11-29:
1.21 Gigawatt?! - wie konnte ich nur so Leichtsinnig sein!

marq24, 2024-11-29:
please update to https://github.com/marq24/ha-goecharger-api2/releases/tag/2024.11.2 and close if fixed

closed 2024-12-06 by marq24