marq24/ha-goecharger-api2 #31: Don't discharge battery

traffiqq, 2024-11-29:
### Description

Hi,

thanks for the good work !
I have a feature request or maybe i am just using the integration wrong.
I have a battery and post the charge/discharge current to pakku.
Is there any way to not use the capacity of the battery to charge the cars ?

marq24, 2024-11-29:
this question must go to go-e ... this integration just is the "messanger"...

closed 2024-11-29 by marq24

traffiqq, 2024-11-29:
@marq24 Yes, you are right. For whom it may concern, i ended up using a template sensor to "hide" the discharge power of the battery unless the SOC is above 90%

`{% if states("sensor.battery_state_of_capacity") | float > 90 %}
  {{states('sensor.power_meter_active_power') | float + states('sensor.battery_charge_discharge_power') | float}}
{% else %}
  {{states('sensor.power_meter_active_power')}}
{% endif %}`

I then used that new sensor in the automation:
`alias: goe charge 11kw pv surplus
description: >-
  Simple automation to provide your go-eChargers with the required data so that
  the wallbox can support PV surplus charging.
triggers:
  - seconds: /5
    trigger: time_pattern
conditions: []
actions:
  - data:
      pgrid: "{{states('sensor.goepowergrid')}}"
      ppv: "{{states('sensor.inverter_input_power')}}"
      pakku: "0"
      configid: Id.ChargerA
    action: goecharger_api2.set_pv_data
  - data:
      pgrid: "{{states('sensor.goepowergrid')}}"
      ppv: "{{states('sensor.inverter_input_power')}}"
      pakku: "0"
      configid: Id.ChargerB
    action: goecharger_api2.set_pv_data
mode: single
`

With this configuration i managed to only use pv surplus and provide power from battery if the SOC is above 90%.

AlexanderTurnowsky, 2025-05-08:
> [@marq24](https://github.com/marq24) Yes, you are right. For whom it may concern, i ended up using a template sensor to "hide" the discharge power of the battery unless the SOC is above 90%
> 
> `{% if states("sensor.battery_state_of_capacity") | float > 90 %} {{states('sensor.power_meter_active_power') | float + states('sensor.battery_charge_discharge_power') | float}} {% else %} {{states('sensor.power_meter_active_power')}} {% endif %}`
> 
> I then used that new sensor in the automation: `alias: goe charge 11kw pv surplus description: >- Simple automation to provide your go-eChargers with the required data so that the wallbox can support PV surplus charging. triggers:
> 
> * seconds: /5
>   trigger: time_pattern
>   conditions: []
>   actions:
> * data:
>   pgrid: "{{states('sensor.goepowergrid')}}"
>   ppv: "{{states('sensor.inverter_input_power')}}"
>   pakku: "0"
>   configid: Id.ChargerA
>   action: goecharger_api2.set_pv_data
> * data:
>   pgrid: "{{states('sensor.goepowergrid')}}"
>   ppv: "{{states('sensor.inverter_input_power')}}"
>   pakku: "0"
>   configid: Id.ChargerB
>   action: goecharger_api2.set_pv_data
>   mode: single
>   `
> 
> With this configuration i managed to only use pv surplus and provide power from battery if the SOC is above 90%.
> 
> Edit: Maybe this isn't even needed. According to the api spec
> 
> pAkku: PvSurPlus information. e.g.: {"pGrid": 1000., "pPv": 1400., "pAkku": 2000.} pGrid < 0 ==> feed grid, pAkku < 0 ==> load battery, pPv > 0 ==> PV production, pPv < 0 ==> standby. Needed all 5 seconds. Can be read back within 10 seconds after set. pPv und pAkku are optional
> 
> May it should suffice if the pAkku power is inverted for my case.

Can you more explain this. I got today my goecharger and i'm interrested how you did this.
At the moment i left out pakku but i think the batterie will be dischared when there is no PV surplus.

My question which sensor did you generated is it sensor.inverter_input_power or the other one.
Can you maybe also share the template sensor configuration from your YAML?

Thanks
Alex