marq24/ha-goecharger-api2 #33: Some entities became

IngmarStein, 2024-12-09:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Some (but not all) entities suddenly became unavailable, but are still available via the API.

Here are some examples:

[image]

I don't know what triggered this:

[image]

because the parameters can still be retrieved via the API, for example http://${CHARGER}/api/status?filter=ama returns `{"ama":16}`

The attached trace may be related.

### Add your DEBUG log output

```shell
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/goecharger_api2/number.py", line 20, in async_setup_entry
    entity = GoeChargerNumber(coordinator, description)
  File "/config/custom_components/goecharger_api2/number.py", line 33, in __init__
    description.native_max_value = 16
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 39, in __setattr__
dataclasses.FrozenInstanceError: cannot assign to field 'native_max_value'
```

IngmarStein, 2024-12-09:
Is it possible that the release tags are wrong?

2024.11.2 seems to contain the the fix for #29 (https://github.com/marq24/ha-goecharger-api2/commit/503e05ed4bfc390a5be5a90f0ddeb685aae73684): https://github.com/marq24/ha-goecharger-api2/compare/2024.11.2...main

But 2024.12.0 does not: https://github.com/marq24/ha-goecharger-api2/compare/2024.12.0...main

IngmarStein, 2024-12-09:
"Downgrading" from 2024.12.0 to 2024.11.2 indeed fixes the issue.

Cillian0305, 2024-12-09:
same issue here, reverted back to 11.2 made all entities available again.
Even more annoying: with 12.0 there are no entities avalable at all if you add a new wallbox.

marq24, 2024-12-09:
Thanks a lot for letting me know @IngmarStein - No real clue what I messed up when I comit/pushed the 2024.12.0 release - in any case there is a new 2024.12.1

IngmarStein, 2024-12-10:
That worked, thanks a lot!

closed 2024-12-10 by IngmarStein