marq24/ha-goecharger-api2 #39: Automatisierung für set_pv_data funktioniert (firmware 5.71)

r00tat, 2025-04-03:
### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hi,
first thanks for the integration and the availability to control the loading state. I tried to use the automation to control loading of my EV, but loading never starts. 
I have followed the steps defined in README.md and configured my Go-E accordingly. 

go-eCharger V4 [11 kW] Local
von go-e GmbH [Austria]
Firmware: 57.1
integration version 2024.12.1

From the logs I can see that the value is pushed fine. If I monitor the value in the UI I can see that the value appears after the set_pv_data is called, but then it switches back to unknown. So it looks like the API first accepts the value, but on the next query it responds with unknown or the value cannot be parsed right. 

This is the overview of the values set through automation:

[image]

At the top you also see "Unbekannt"

(I have set the automation to a fixed value for testing purposes right now:

```yaml
action: goecharger_api2.set_pv_data
metadata: {}
data:
  pgrid: 0
  ppv: 5000
  pakku: 0

```

In the details you can see the toggle of values:

[image]

here are the configured values of my home assistant instance.

```jinja
ppv: {{ states('sensor.goe_208531_ppv')}}
grid: {{ states('sensor.goe_208531_pgrid')}}
batterie: {{ states('sensor.goe_208531_pakku')}}

LMO: {{ states('select.goe_208531_lmo')}}
fup: {{states('switch.goe_208531_fup')}}
acp: {{states('switch.switch.goe_208531_acp')}}
```

```yaml
ppv: unknown
grid: unknown
batterie: unknown

LMO: 4
fup: on
acp: unknown
```
I have also configured a "Flexibler Energietarif", even I do not use it.

I noticed that Status shows "Nicht laden, weil 'Fallback (günstiger Strompreis)'". This might indicate some error in the software. 

I know that this might be an issue related to the firmware, but I wanted to share this details as a heads up already. 

### Add your DEBUG log output

```shell
2025-04-03 20:17:38.732 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Initialized trigger Go-e Charger PV Überschuss
2025-04-03 20:17:40.863 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Go-e Charger PV Überschuss: Running automation actions
2025-04-03 20:17:40.863 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Go-e Charger PV Überschuss: Executing step call service
2025-04-03 20:17:40.864 DEBUG (MainThread) [custom_components.goecharger_api2] Provided config_id: None -> using goecharger_api2 serial: 208531
2025-04-03 20:17:40.864 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': 0.0, 'pPv': 5000.0, 'pAkku': 0.0}
2025-04-03 20:17:40.864 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": 0.0, "pPv": 5000.0, "pAkku": 0.0}'} to go-eCharger@http://192.168.1.x
2025-04-03 20:17:40.919 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': 0.0, 'pPv': 5000.0, 'pAkku': 0.0}}
2025-04-03 20:18:03.701 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 58 keys from go-eCharger@http://192.168.1.x
2025-04-03 20:18:03.769 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.1.x
2025-04-03 20:18:03.769 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.069 seconds (success: True)
2025-04-03 20:18:07.851 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Initialized trigger Go-e Charger PV Überschuss
2025-04-03 20:18:10.400 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Go-e Charger PV Überschuss: Running automation actions
2025-04-03 20:18:10.400 INFO (MainThread) [homeassistant.components.automation.go_e_charger_pv_uberschuss] Go-e Charger PV Überschuss: Executing step call service
2025-04-03 20:18:10.401 DEBUG (MainThread) [custom_components.goecharger_api2] Provided config_id: None -> using goecharger_api2 serial: 208531
2025-04-03 20:18:10.401 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': 0.0, 'pPv': 10000.0, 'pAkku': 0.0}
2025-04-03 20:18:10.401 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": 0.0, "pPv": 10000.0, "pAkku": 0.0}'} to go-eCharger@http://192.168.1.x
2025-04-03 20:18:10.463 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': 0.0, 'pPv': 10000.0, 'pAkku': 0.0}}
2025-04-03 20:18:33.702 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 58 keys from go-eCharger@http://192.168.1.x
2025-04-03 20:18:33.793 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.1.x
2025-04-03 20:18:33.794 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.092 seconds (success: True)
```

marq24, 2025-04-04:
You must send **PGRID** data > 0 

your log shows `write result: {'ids': {'pGrid': 0.0, 'pPv': 10000.0, 'pAkku': 0.0}}` 
https://github.com/marq24/ha-goecharger-api2?tab=readme-ov-file#pgrid-required

have you double check, that the charger IS actually receiving the data?
https://github.com/marq24/ha-goecharger-api2?tab=readme-ov-file#finally-verify-if-the-wallbox-receive-your-data-from-the-automation

also important:
https://github.com/marq24/ha-goecharger-api2?tab=readme-ov-file#do-not-forget-this-important-settings

r00tat, 2025-04-04:
Hi Marq24,

according to the docs it should be a negative value, if power is available to be consumed by GoE. 

I have tried using positive and negative values for pgrid, but still no luck. 

For this settings: 

```yaml
action: goecharger_api2.set_pv_data
metadata: {}
data:
  pgrid: -4000
  ppv: 5000
  pakku: 0
response_variable: result
```

I see the following logs:

```
2025-04-04 21:35:10.194 DEBUG (MainThread) [custom_components.goecharger_api2] Provided config_id: None -> using goecharger_api2 serial: 208531
2025-04-04 21:35:10.194 DEBUG (MainThread) [custom_components.goecharger_api2] Service set PV data: {'pGrid': -4000.0, 'pPv': 5000.0, 'pAkku': 0.0}
2025-04-04 21:35:10.194 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to write {'ids': '{"pGrid": -4000.0, "pPv": 5000.0, "pAkku": 0.0}'} to go-eCharger@http://192.168.1.21
2025-04-04 21:35:10.244 DEBUG (MainThread) [custom_components.goecharger_api2] write result: {'ids': {'pGrid': -4000.0, 'pPv': 5000.0, 'pAkku': 0.0}}
2025-04-04 21:35:10.257 INFO (MainThread) [custom_components.goecharger_api2.automation] Go-Echarger set_pv result {"date": "21:35:10.255018", "response": {"ids": {"pAkku": 0.0, "pGrid": -4000.0, "pPv": 5000.0}}, "success": "true"}
```

I have added a diagram and the relevant entities to the dashboard. The statistics diagram shows some values for pgrid aso, but the entitiy cards do not:

[image]

The App still shows "PV-Überschuss nicht verfügbar"

[image]

I have also tested with positive grid values, but still the same outcome. 

Also I followed your guide and configured the values accordingly and the Engergietarif. 

[image]

r00tat, 2025-04-04:
I have tried a downgrade to firmware 56.2 but this broke the wifi connection. 

In the release notes of 57.1 there is something interesting:

"Set Energy Limit in Homeassistant hat nun korrekte Parameter"

Is this change related to your integration?

BTW we can switch to German, if you'd like ;-)

r00tat, 2025-04-04:
I noticed something on apikeys docs: _Needed all **5 seconds.** Can be read back within 10 seconds after set. pPv und pAkku are optional_

My Automation was running all 10 secs or even less, as I wanted to reduce the load on HA.

This now leads to the values being displayed:

[image]

I was able to verify that behaviour by running two simple scripts to set and read the pgrid, ppv, and pakku values. 

So indeed it was my misconfiguration and not your integration. 

Maybe it is worth mentioning in the README, that the automation is required to run all 5 seconds. 

Thanks for the integration!

marq24, 2025-04-05:
IMHO all is there...

https://github.com/marq24/ha-goecharger-api2?tab=readme-ov-file#finally-verify-if-the-wallbox-receive-your-data-from-the-automation

[image]

closed 2025-04-05 by marq24