marq24/ha-goecharger-api2 #40: "Last button press" is always null

herrfrei, 2025-04-19:
### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I'd like to use the lbp status value (last button press) for further automations in HA. But the value is not submitted (always null). What is needed to activate the transmission?

### Add your DEBUG log output

```shell

```

marq24, 2025-04-19:
IF you have understood the Shit IN — Shit OUT principle, then I really wonder why yiu created this issue here... since at least for me its obvious, that the vale is currently not provided by the API... and the issue in the goe github already exists.

marq24, 2025-04-19:
... could it be, that "last button pressed" is referring to the physical button on the charger itself ?!

herrfrei, 2025-04-19:
I expect that. So I pressed the button (single and double press), but lbp
is always null.

2025, 21:49:

marq24, 2025-04-19:
In any case, when the go-eCharger API return's null, then there is nothing the Integration could do about that...

herrfrei, 2025-04-19:
Okay, many thanks!

2025, 21:58:

closed 2025-04-20 by marq24