marq24/ha-goecharger-api2 #41: Dynamic charging limits to charger from Home Assistant

jkkataja, 2025-04-20:
### Description

I tried googling around a bit but did not find answers. Is it currently possible to use power consumption sensors to feed dynamic charging limits to a charger by essentially emulating the controller functionality?

I would assume that it would be done in similar fashion as data is fed currently to enable PV surplus charging, but have the fields required been spotted by anyone?

Did also have look at the API keys (https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-en.md ) but did not spot anything that would be immediately clear to me at least.

### API Keys

```shell

```

marq24, 2025-04-20:
> power consumption sensors to feed dynamic charging limits to a charger

you might need to explain this a bit more in detail - but in general there is no documentation available (beyond the PV surplus charging options (data that can be send to the charger)) - of course you can send "any fake PV data" to the changer to pretend, that there is energy that can be used...

jkkataja, 2025-04-20:
That’s what I was afraid of that there wouldn’t be any documentation for this.

But to explain in more detail what I’m referring to what’s described here https://go-e.com/en/magazine/what-is-dynamic-load-balancing

TLDR: If a phase is exceeding the main fuse limit, the charger will reduce the power draw to keep the power usage in a safe range.

Since it’s basically the same thing as what we are doing with excess PV charging I’d guess it’s only the matter of being able to figure out the necessary parameters to be send to charger, I wonder if Go-E people will be willing to disclose it 🤔

Another option is, like you said, using the PV surpluss with adjusted values.

marq24, 2025-04-21:
If we are talking about single phase loading, then you can create a simple automation observing your home consumption and based on this value you can adjust the `number.goe_[serial]_amp` ?!

jkkataja, 2025-04-22:
The automation should probably calculate the available current per phase and then adjust that value based on what the minimum available current.

So for each phase we need the logic to determine if we are exceeding the main fuses and if yes, how much we need to reduce the current or if we need to stop the charging altogether. I do understand if you want to close this issue as it starting to be on the borderline if it's in scope or not. I could try to come up with automation that would solve this.

marq24, 2025-04-22:
I hope its reasonable for you, that I can't assist here further (with creating such an automation) - But of course I would share it here as part of the documentation if that's fine for you.

jkkataja, 2025-05-12:
I've been really busy recently and since I won't be able to create this in a reasonable timeline I'll close this issue.

I did receive information from go-e that the static load balancing should be used so in effect adjusting the value marq24 mentioned earlier.

closed 2025-05-12 by jkkataja