marq24/ha-goecharger-api2 #44: "Allow car pause" setting requires a restart of the charger to accept changes

NikolajChristensen, 2025-05-09:
Hi there,

In my setup "allow car pause" is indeed required in order for the charger to pause charging once PV surplus drops below the limit.

However if I turn "allow car pause" off, and then on again, the charger will not pause charging due to low PV surplus until I have rebooted the charger.

Not sure if this is related to the "won't stop charging" issues outlined in the documentation. But perhaps it is?

Also this could be a bug for go-e - not the integration. Let me know what you think. Thanks!

And thanks for an awesome integration.

marq24, 2025-05-09:
Hi,

this indeed sounds like an general issue of the charger - please be so kind and verify IF the behavior is identical when using the mobile app to make similar adjustments.

NikolajChristensen, 2025-05-09:
Hm I would if I could find the darn setting in the mobile app :)

NikolajChristensen, 2025-05-10:
Any hints as to where "enable car pause" is in the app?

marq24, 2025-05-10:
I also searched in the 'Konfiguration des Lagevorgangs' -> "Auto"... but no toggle will switch the `acp` flag :-/ so I have to admit that I am clueless...

marq24, 2025-05-10:
... on my charger this `acp` is set to `true`... (jfyi)

NikolajChristensen, 2025-05-10:
Thanks. I have reached out to go-e support for now. We'll see.

NikolajChristensen, 2025-05-31:
Go-E support is looking into it. I sense it's not high priority - maybe that's ok too.

closed 2025-05-31 by NikolajChristensen