marq24/ha-goecharger-api2 #45: Device cannot be added to Dashboard.

Eselsteve, 2025-05-21:
### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Device cannot be added to Dashboard.

### Add your DEBUG log output

```shell

```

marq24, 2025-05-21:
please provide more details what is really the issue

Eselsteve, 2025-05-21:
Normally, any new devices I add appear automatically in the dashboard or I add them separately to the dashboard, but that doesn't work here.

Eselsteve, 2025-05-21:
Guten Abend übrigens 😎

Eselsteve, 2025-05-21:
A screenshot won't help because, as I said, ALL other added devices automatically appear in the dashboard. But not here. The "Add to Dashboard" button doesn't work either.

marq24, 2025-05-22:
see - you want help - then it would be highly appreciated if you would be provide the information that I am asking for - since I am __not able to reproduce your 'reported' issue__ here locally - so I just can assume, that I don't understand the procedure (step by step) that you are doing in your HA installation... 

There are multiple different "Add to Dashboard" buttons in the HA UI - all of the ones I know work rock solid here.

closed 2025-05-23 by marq24