marq24/ha-goecharger-api2 #50: Phase wish-mode is always showing Force 3-Phases, regardless of mode change

nbuddy, 2025-06-14:
### Checklist

- [x] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I understand the SHIT-__IN__ - SHIT-__OUT__ principle (in case of implausible data) | It will happen, that the data that is displayed by this integration does not make much sense (to you) - aka the data is not plausible. __Of course__ it could be the case, that something in this integration has been messed up - but so far - in all reported issues the root cause of implausible data was/is, that the go-eCharger device itself already provided this data [you can check this by directly requesting the attribute from the wallbox]
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

After a fresh install of go-e integration in HA, I noticed the following issue:

In Diagnostic parameters, "Phase wish-mode" and "Phase wish-mode [CODE]" are ALWAYS showing the following status, regardless of changing the "Phase Switch-Mode" setting, either via integration or via go-e mobile app and the actual mode:
- Phase wish-mode: Force 3-Phases
- Phase wish-mode [CODE]: 0

How to reproduce the issue:
1. In HA Integrations, go to "go-e APIv2 Connect", device configuration page
2. Change the "Phase Switch-Mode" parameter between "Auto", "Force one phase only" and "Force three phases"
3. In the go-e mobile app, verify successful mode change (almost immediate)
4. In the HA "go-eCharger API v2 [nnn] Local", scroll down to "Diagnostic" info
5. Notice that the "Phase wish-mode" is always showing "Force 3-Phases" and "Phase wish-mode [CODE]" is always showing "0", regardless of mode change

Details of setup:
- HA Core: 2025.6.1
- go-e Hardware version: V4
- go-e Firmware: 57.1
- go-e connection: Local network

Debug info:
- Enabled debug logging and restarted HA
- Switched through all 3 modes with "Phase Switch-Mode"
- Observed "Phase Switch-Mode", which remained the same despite the mode change
- Exported debug log

Screenshots and debug log attached.

[image]

### Add your DEBUG log output

```shell
2025-06-14 10:48:36.093 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration goecharger_api2 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-06-14 10:48:40.041 INFO (MainThread) [custom_components.goecharger_api2] 
-------------------------------------------------------------------
go-e APIv2 Connect
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/marq24/ha-goecharger-api2/issues
-------------------------------------------------------------------

2025-06-14 10:48:40.041 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:48:40.099 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/__init__.py:2: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  __import__('pkg_resources').declare_namespace(__name__)

2025-06-14 10:48:41.634 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/ProtocolMessage.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(

2025-06-14 10:48:41.635 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/AudioFadeMessage.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(

2025-06-14 10:48:41.636 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/PlayerPath.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(

2025-06-14 10:48:41.637 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/Origin.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(

2025-06-14 10:48:41.638 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/DeviceInfoMessage.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
  warnings.warn(

2025-06-14 10:48:41.639 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.1 is exactly one major version older than the runtime version 6.31.1 at pyatv/protocols/mrp/protobuf/Common.proto. Please update the gencode to avoid compatibility violations in the next runtime release.
[... 145 lines cut ...]
2025-06-14 10:51:16.348 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:51:16.464 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.50.111
2025-06-14 10:51:16.473 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.126 seconds (success: True)
2025-06-14 10:51:21.346 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:51:21.478 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.50.111
2025-06-14 10:51:21.478 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.132 seconds (success: True)
2025-06-14 10:51:26.346 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:51:26.512 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.50.111
2025-06-14 10:51:26.512 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.167 seconds (success: True)
2025-06-14 10:51:31.346 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:51:31.629 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.50.111
2025-06-14 10:51:31.629 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.283 seconds (success: True)
[... cut ...]
2025-06-14 10:51:51.345 DEBUG (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] going to request 60 keys from go-eCharger@http://192.168.50.111
2025-06-14 10:51:51.498 INFO (MainThread) [custom_components.goecharger_api2.pygoecharger_ha] [missing fields: 3 -> ['ccu', 'rfb', 'rst']] - not all requested fields where present in the response from from go-eCharger@http://192.168.50.111
2025-06-14 10:51:51.498 DEBUG (MainThread) [custom_components.goecharger_api2] Finished fetching goecharger_api2 data in 0.153 seconds (success: True)
```

marq24, 2025-06-14:
First of all Thank you very much for this detailed Issue report... I must prepare the pool now... So I can not promise to get into the details in the next 8h

nbuddy, 2025-06-16:
After discussed with go-e support, it turned out that I had misunderstood the meaning of **pwm** attribute in first place.

Here's the response from go-e support:

_The phase wish mode (pwm) doesn’t take the phase switching into account. 
It only shows what the go-e Charger would do in case of PV Surplus charging, which means:

- Charger is in Eco mode with PV surplus activated and currently charging

- There’s enough surplus for the charger to only charge single-phased → pwm will be 1 (Wish 1 = Charger would like to be in 1-phase charging right now).

- The PV-production raises to e.g. 7kW and so does the surplus raise. Now there’s even enough PV-surplus for the charger to switch to 3-phases → pwm will change to 2 (Wish 3 = Charger would like to be in 3-phase mode now because the conditions allow it)

The key itself doesn’t have anything to do with you switching manually from 3-phases to 1-phase (or the other way around)._ 

So I'd consider this case is closed.

Thank you @marq24 for checking out anyway!

closed 2025-06-16 by marq24