marq24/ha-goecharger-api2 #62: Change Power Preference via HA

achmet9948, 2025-08-14:
### Description

Hello, is it possible to set up the power preference via Homeassistant? 
It would be great to implement this, because I want to create an automation, that switch the preference, if the home battery below 15 % or between 15  and 60%.
At the moment the battery is not loading, when surpluse loading is on. 

### API Keys

```shell

```

marq24, 2025-08-14:
So you want to use PV power available to charge Battery first... Then IMHO you should modify the value (feed into the grid) you send to the wallbox... So there is enough energy left that will be send to your Battery?

marq24, 2025-08-14:
... I am not sure, if you understood my reply correctly - _you_ can / should do that yourself.

achmet9948, 2025-08-14:
Yes I did not understand you. Sorry

I think, it should be possible direct via an entity. On the goe app you can change the power preferences directly. 
But I found the key now (frm). This is only a read API. I will send a feature request to goe. Thanks a lot

closed 2025-08-14 by achmet9948